sipeed/MaixPy-v1 #428: 該如何在MaxiPy IDE 上 import tensorflow？

KuoEuran, 2021-08-14:
**Describe the bug**
該如何使用tensorflow 的API?目前是無法import tensroflow

**Expected behavior**
tensorflow的API可以使用

**Actual behaviour**
無法使用tensorflow的function

**Screenshots**
[image]
**Please complete the following information**
 - IDE version: [Maixpy 1.6.7]
 - Firmware version: [0.6.2 latest firmware]
 - Board: [Maix Arduino]
 - OS: [e.g. Windows10]

再麻煩幫忙了謝謝

Neutree, 2021-08-14:
不支持

closed 2021-08-14 by Neutree

KuoEuran, 2021-08-14:
@Neutree 您好,因為我在https://blog.csdn.net/qq_36229876/article/details/105674765的部落格上有看到他可以使用tf.image.non_max_suppression的API,但我目前是不知道怎麼使用,請問要用nms的話要如何使用呢?