sipeed/MaixPy-v1 #429: error no 12: ENOMEM even running helloworld.py

6W3N, 2021-08-18:
Hi, 
I'm a newbie with Sipeed, and I want to perform sound classification and sound localization using Maix Bit and MaixPy IDE.
For this, I had run the two examples on [here](https://en.bbs.sipeed.com/t/topic/683/2) and [here](https://elchika.com/article/af968c98-a6d0-41fa-8e09-291c43eddd40/). The former is a script for trained classification using tiny-yolo, and the latter is for sound localization using a microphone array.
After downloading the latest firmware (*_with_lvgl.bin) and 20class.kfpkg, I succeeded in the 20class sample (the former sample). But I did not perform the latter sample with the error no. 12: ENOMEM.

I read the [issue about the ENOMEM](https://github.com/sipeed/MaixPy/issues/243), and I understood that this error shows a memory limitation. So we understood this error rise when I run the specific program (here, the latter sample). But I receive the same error message even running helloworld.py (It was working before!). Additionally, I receive the same error whenever the command `sensor.set_framesize(sensor.QVGA)` is used.

What imitates this situation? How can I solve this problem?

Many thanks,

System:
- Ubuntu 20.04
- MaixPy IDE 0.2.5
- Firmware 0.6.2_59
- Maix Bit with mic
- Sipeed R6+1 Microphone Array

Neutree, 2021-08-19:
maybe you can use the minimum version or standard version,(download here: https://dl.sipeed.com/shareURL/MAIX/MaixPy/release/master) , the memory of k210 only 6MiB, firmware size + model size + program memory usage(like micropython gc heap, and camera buffer lcd buffer etc.) must < 6MiB

6W3N, 2021-08-19:
Thanks for the prompt reply.
I got it, but the next question arises.
How can I remove the old firmware or model or program? Can I use the erase function of the kflash GUI?

Thanks,

Neutree, 2021-08-19:
just update new firmware, will cover the old one, and you can also erase first

6W3N, 2021-08-23:
Sorry for the late reply.
I was able to delete the old .kfpkg and run helloworld.py.

closed 2021-08-23 by 6W3N