sipeed/MaixPy-v1 #434: kmodel v5 support

oplik0, 2021-09-07:
Current versions of nncase are already generating kmodel v5 files, but it seems it's not yet supported by MaixPy - calling KPU.load on a v5 file seems to throw random errors (see #423 #426 and #433 for examples), or sometimes just crash the firmware with no errors even reported.

So currently one needs to either use v0.2 nncase (which lacks some ops and doesn't support YoloX) or use Kendryte SDK instead of MaixPy.

Narduzzi, 2021-12-06:
Any update on this?

cscribano, 2022-11-08:
Adding support for kModelV5 would be extremely appreciable.
Sadly, the required libraries in [components](https://github.com/sipeed/MaixPy/tree/master/components)/[kendryte_sdk](https://github.com/sipeed/MaixPy/tree/master/components/kendryte_sdk/libs) are closed source.

cscribano, 2022-11-08:
Update: apparently canaan is actively developing a derivative of MaixPy which should solve the issue: https://github.com/kendryte/canmv
I successfully built and flashed on MaixDock, the provided examples work just fine, kModelsV5 should be supported!!