sipeed/MaixPy-v1 #435: no video support in maixpy_v0.6.2_68_g520879e06_openmv_kmodel_v4_with_ide_support.bin

easy-and-simple, 2021-09-20:
I didn't test other builds, but if this build have problems all build should be rebuild

easy-and-simple, 2021-09-22:
Guys I build few different configurations to see how modules impact binary file size and found that video module doesn't impact final file size, so I can't understand why it is not included in build I specified above.
I build image that includes support for v4, video and speech with minimum openmv and size of bin file is also 1.4MB

Neutree, 2021-09-22:
you can manually compile MaixPy project, and select modules you need, for example video:

[image]

video module not occupy too much memory, there's the compare of include video or not
[image]

easy-and-simple, 2021-09-22:
If you don't need customers, then your replay is normal

easy-and-simple, 2021-09-22:
Image that I build doesn't support board info except boot key

Neutree, 2021-09-23:
board info has been changed by @junhuanchen , doc here: https://wiki.sipeed.com/soft/maixpy/zh/get_started/board_info.html