sipeed/MaixPy-v1 #436: как работает?

Yabuku0, 2021-09-21:
Добрый день. Недавно стал пользователем платы Maixduino. Прошу помощи в вопросах общеобразовательных. 1) Как формируется (создается), можно ли где нибудь посмотреть ее код, узнать что внутри. 2) Что за параметры train и valid? 3) все ли нормально на данном отчете? просто данная модель не распознает объекты вообще. 4) и самое главное можете дать ссылки на инструкции, документацию по maixduino и в общем по тематике микроконтроллеров и их программирования, машинного зрения, обучения что нибудь фундаментальное. Спасибо
[image]
)

Yabuku0, 2021-09-21:
> Добрый день. Недавно стал пользователем платы Maixduino. Прошу помощи в вопросах общеобразовательных. 1) Как формируется (создается), можно ли где нибудь посмотреть ее код, узнать что внутри. 2) Что [...]

Good afternoon. Recently became a user of Maixduino board. I ask for help in matters of general education. 1) How is it formed (created), is it possible to look at its code somewhere, find out what is inside. 2) What are the train and valid parameters? 3) is everything normal on this report? it's just that this model does not recognize objects at all. 4) and most importantly, you can give links to instructions, documentation on maixduino and, in general, on the topic of microcontrollers and their programming, machine vision, learning something fundamental. thanks

easy-and-simple, 2021-09-22:
А ты не слишком борзый? Сюда такие вопросы не задаются
Для этого есть Яндекс. Если искать не умеешь вот тебе прямая ссылка
https://yandex.ru/search/?text=maixpy&lr=10379&clid=2270456&win=490
Идешь по очереди. Второй результат тебе сюда привел, читай всех остальных. Там написано все что тебе интересует.
Что такое трейн и валид поймешь когда научишь как все это работает

Neutree, 2021-09-22:
@Yabuku0 you can find train code here: https://github.com/sipeed/maix_train

and docs here: https://wiki.sipeed.com/soft/maixpy/zh/index.html , and you can use translate 
[image]