sipeed/MaixPy-v1 #437: I use YOLO to inspect car images. But I don't know how to count car pictures yet. Help me at

Name1112, 2021-09-22:
I am now able to detect images of the car. by drawing up a frame But I don't know how to use this frame to be able to count frames. Help me at

Code:
import sensor,image,lcd,time
import KPU as kpu
from Corgi85 import corgi85
import random

lcd.init()
lcd.rotation(2)

sensor.reset()
sensor.set_pixformat(sensor.RGB565)
sensor.set_framesize(sensor.QVGA)
sensor.set_vflip(0)
sensor.set_hmirror(0)
sensor.run(1)

classes = ['aeroplane', 'bicycle', 'bird', 'boat', 'bottle', 'bus', 'car', 'cat', 'chair', 'cow', 'diningtable', 'dog', 'horse', 'motorbike', 'person', 'pottedplant', 'sheep', 'sofa', 'train', 'tvmonitor']
task = kpu.load(0x500000)
a = kpu.set_outputs(task, 0, 7,7,35)
anchor = (1.889, 2.5245, 2.9465, 3.94056, 3.99987, 5.3658, 5.155437, 6.92275, 6.718375, 9.01025)
a = kpu.init_yolo2(task, 0.5, 0.3, 5, anchor)
clock = time.clock()

while(True):
    clock.tick()
    img = sensor.snapshot()
    setto = kpu.run_yolo2(task, img)

    if setto:
        for i in setto:

            if i.classid() == 6 : 
                a=img.draw_rectangle(i.rect(),color = (0, 255, 0),thickness=4)
                a=img.draw_string(i.x(), i.y(), classes[i.classid()], color=(0, 255, 0),scale=2)

    lcd.display(img)

a = kpu.deinit(task)

easy-and-simple, 2021-09-22:
This is not a forum.
Ask your question  in sipeed forum

Name1112, 2021-09-22:
Can you tell me how to count? I'm just starting to study so I don't understand a lot. Thank you.

UncleGrumpy, 2021-09-22:
Cars = 0

…
…

if i.classid() == 6 :
    Cars += 1
    …
    …

easy-and-simple, 2021-09-22:
You can't understand even that this is not forum! Maybe you should go to school at first to learn some basics?!