sipeed/MaixPy-v1 #444: /sd not linking to SD card mount

andredankert, 2021-11-22:
Had issues with recording videos and images (https://github.com/sipeed/MaixPy/issues/442) and experimented a bit. Figured I see sdcard content and the new files, but it is not on the physical card. Tried reflashing FW and even erasing whole chip (because it filled up with files). Now I see the "virtual SD card" running
`import os
print(os.listdir("\\sd"))`
Output: `['freq.conf', 'main.py', '\\sd\\vid1.avi', '\\sd\\vid.avi']`
in the IDE. If I run `os.listdir("\\sd")` in picocom I actually get
`['main.py']`
,which is the actual physical SD card content. 
First, how can it be different. And why are all my IDE scripts accessing this "virtual" SD instead?

**Please complete the following information**
 - IDE version: 0.2.5
 - Firmware version: 0.6.2
 - Board: Maix Bit
 - OS: Ubuntu Linux

arthurkafer, 2022-02-24:
I think the default folder that the board starts is /flash, but you can find SD files using /sd

I don't know if you already tried it

andredankert, 2022-02-25:
It was quite random, sometimes it worked, sometimes it didn't. It's a while ago, I don't remember what I did, but in the end it was more reliable and I could work with it. It was just weird that in the same startup cycle picocom and IDE showed different outputs for the same command!