sipeed/MaixPy-v1 #447: Doubts regarding boot.py

abhi-84, 2021-11-30:
Hi all
Let;s assume I have transferred code to the board which is placed in boot.py. Now, How can I replace my code in boot.py. If i again send new code, will it overwrite the previous code or extend the boot.py file?
What OS is running by default which is managing filesystem and other behavior. What is uos module?

Please guide

robert-hh, 2021-11-30:
If you transfer boot.py again, the old file will be replaced.
The file system will be managed by the MicroPython firmware. 
uos is a "micro" version of the Python os module.

abhi-84, 2021-11-30:
Thank you for responding and providing clarification.
Can you also tell me about the usage of FreeRTOS in Maixpy? Apart from providing os related functionalities at the application level such as queues, semaphores, tasks, etc does it have any other role at the board level? Can freertos be used along with AI applications?