sipeed/MaixPy-v1 #455: new user: how to install new MaixPy firmware

dylanmc, 2022-01-27:
**Describe the bug**
I have a MAiX Amigo, it powers on and reports firmware v0.5.0. I have downloaded a new firmware, and tried copying it to a micro-SD card and booting the Amigo, but nothing I do (holding down different buttons, etc.) seems to cause a firmware update. Unclear from the website how to get started.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to https://github.com/sipeed/MaixPy/releases
2. Download latest firmware
3. Copy .bin file to FAT-32 formatted MicroSD card, insert into Amigo
4. Power on (with xyz button pressed)

**Expected behavior**
Firmware update happens

**Actual behaviour**
Screen reports "Welcome to MaixPy v0.5.0 : maixpi.sipeed.com"

**Screenshots**

**Please complete the following information**

 - Firmware version: 0.5.0
 - Board: Maix Amigo
 - OS: n/a

dylanmc, 2022-02-06:
I figured it out (in case others have issues). 

Steps:
 * download this repository, to get the tools and scripts needed to flash the firmware
 * cd MaixPy/projects/maixpy_amigo_tft (or whichever board you have)
 * plug in your MaixPy device
 * figure out which serial device it has created (in my case, on macOS, it created `/dev/tty.usbserial-004043B`) it also created an `...A` device, but that didn't seem to with the flash command
 * download the [maixpy firmware release](https://github.com/sipeed/MaixPy/releases) you want to flash, and move it to `./build/maixpy.bin`
 * run the flash command of the `project.py` script:
    `python3 project.py -B maixduino -p /dev/tty.usbserial-004043B -b 1500000 flash`
 * this script flashes the file in build/maixpy.bin to your device. 

You should get output like this:
```
-- SDK_PATH:/Users/.../MaixPy
maixpy
-- flash config changed, update at /Users/.../MaixPy/projects/maixpy_amigo_tft/.flash.conf.json
-- flash start
=============== flash config =============
-- flash port    :/dev/tty.usbserial-004043B
-- flash baudrate:1500000
-- flash board:maixduino
-- flash open terminal:False
-- flash download to sram:False
-- flash noansi:False
-- flash slow mode:True
-- flash firmware:/Users/.../MaixPy/projects/maixpy_amigo_tft/build/maixpy.bin

-- kflash start
[INFO]COM Port Selected Manually: /dev/tty.usbserial-004043B
[INFO]Default baudrate is115200, later it may be changed to the value you set.
[INFO]Trying to Enter the ISP Mode...
*
[INFO]Greeting Message Detected, Start Downloading ISP
Downloading ISP: |===============================================================================| 100.0% 10kiB/s
[INFO]Booting From 0x80000000
[INFO]Wait For 0.1 second for ISP to Boot
[INFO]Boot to Flashmode Successfully
[INFO]Selected Baudrate: 1500000
[INFO]Baudrate changed, greeting with ISP again ...
[INFO]Boot to Flashmode Successfully
[INFO]Selected Flash: On-Board
[INFO]Initialization flash Successfully
Programming BIN: |===============================================================================| 100.0% 54kiB/s
[INFO]Rebooting...
== flash end ==
```

Caveat, after following these instructions, my amigo isn't showing anything on its display, nor creating a virtual TTY device, so I can't reflash, so I've got a completely different problem to solve now, but the above instructions might help folks flash their devices.

arthurkafer, 2022-02-24:
Every MaixPy firmware comes with a boot.py and a main.py file, but it probably doesn't have anything in the default file using LCD. 
You can flash AI models just like you flashed your firmware, but it has some other steps to follow to make it work correctly