sipeed/MaixPy-v1 #456: ValueError: [MAIXPY]kpu: region_layer_init err!

arthurkafer, 2022-02-23:
**Describe the bug**
I'm using a simple example with a custom trained yolov2 model with 10 anchors, but after running one time `kpu.run_yolo2(task,img)`, the error `ValueError: [MAIXPY]kpu: region_layer_init err!` occours.

Here's the code I'm using:

```
import sensor, image, time
import KPU as kpu

sensor.reset()					  # Reset and initialize the sensor. It will run automatically, call sensor.run(0) to stop
sensor.set_pixformat(sensor.RGB565) # Set pixel format to RGB565 (or GRAYSCALE)
sensor.set_framesize(sensor.QVGA)   # Set frame size to QVGA (320x240)
sensor.skip_frames(time = 2000)	 # Wait for settings take effect.
sensor.set_windowing((224,224))
clock = time.clock()				# Create a clock object to track the FPS.

kpu.memtest()
task = kpu.load(0x00300000)
kpu.init_yolo2(task, 0.2, 0.05, 10, (0.5081,1.1568,0.7518,1.9968,0.7652,1.3357,1.075,2.5246,1.1645,1.6786,1.6200,2.9711,1.7510,2.1547,2.6649,2.6007,2.7243,6.5252,4.6151,6.6990))
kpu.set_outputs(task, 0, 7, 7, 60)
kpu.memtest()
time.sleep(1)

while(True):
	clock.tick()					# Update the FPS clock.
	img = sensor.snapshot()		 # Take a picture and return the image.
	code = kpu.run_yolo2(task, img)
	if code:
		print(code)
	print(clock.fps())
	kpu.memtest()
```
Here's the model I'm using, it's a simple person detector with one label ("person"):

[model.zip](https://github.com/sipeed/MaixPy/files/8126966/model.zip)
it's zipped, but it's a kmodel format.

I have a similar 5 anchors model that works fine, but I would like to use the 10 anchors one because it is better at recognizing in general.

**Expected behavior**
I expected that the code would continue running in the while loop

**Actual behaviour**
It runs one time and than the Exception occours

**Screenshots**
[image]
```
[MAIXPY] Pll0:freq:806000000
[MAIXPY] Pll1:freq:398666666
[MAIXPY] Pll2:freq:45066666
[MAIXPY] cpu:freq:403000000
[MAIXPY] kpu:freq:398666666
[MAIXPY] Flash:0xef:0x17
[MaixPy] gc heap=0x80133ae0-0x801b3ae0(524288)
[MaixPy] init end

 __  __              _____  __   __  _____   __     __
|  \/  |     /\     |_   _| \ \ / / |  __ \  \ \   / /
| \  / |    /  \      | |    \ V /  | |__) |  \ \_/ /
| |\/| |   / /\ \     | |     > <   |  ___/    \   /
| |  | |  / ____ \   _| |_   / . \  | |         | |
|_|  |_| /_/    \_\ |_____| /_/ \_\ |_|         |_|

Official Site : https://www.sipeed.com
Wiki          : https://maixpy.sipeed.com

init i2c:2 freq:100000

[MAIXPY]: find ov2640
[MAIXPY]: find ov sensor
###free gc heap memory : 496 KB

###free sys heap memory: 3996 KB

###free gc heap memory : 496 KB

###free sys heap memory: 724 KB

[{"x":73, "y":92, "w":44, "h":30, "value":0.250000, "classid":5, "index":0, "objnum":16}, {"x":89, "y":133, "w":45, "h":30, "value":0.250000, "classid":1, "index":1, "objnum":16}, {"x":184, "y":116, "w":47, "h":30, "value":0.284483, "classid":1, "index":2, "objnum":16}, {"x":0, "y":169, "w":38, "h":31, "value":0.420219, "classid":4, "index":3, "objnum":16}, {"x":2, "y":42, "w":27, "h":29, "value":0.411482, "classid":4, "index":4, "objnum":16}, {"x":130, "y":45, "w":27, "h":29, "value":0.477112, "classid":4, "index":5, "objnum":16}, {"x":57, "y":93, "w":45, "h":28, "value":0.250000, "classid":1, "index":6, "objnum":16}, {"x":185, "y":97, "w":45, "h":29, "value":0.267539, "classid":4, "index":7, "objnum":16}, {"x":89, "y":134, "w":45, "h":28, "value":0.351846, "classid":4, "index":8, "objnum":16}, {"x":141, "y":182, "w":45, "h":32, "value":0.454585, "classid":5, "index":9, "objnum":16}, {"x":192, "y":187, "w":31, "h":32, "value":0.277362, "classid":3, "index":10, "objnum":16}, {"x":1, "y":42, "w":29, "h":30, "value":0.392857, "classid":0, "index":11, "objnum":16}, {"x":129, "y":46, "w":29, "h":30, "value":0.464286, "classid":0, "index":12, "objnum":16}, {"x":184, "y":97, "w":47, "h":29, "value":0.250000, "classid":0, "index":13, "objnum":16}, {"x":87, "y":133, "w":49, "h":30, "value":0.321429, "classid":0, "index":14, "objnum":16}, {"x":0, "y":170, "w":39, "h":30, "value":0.392857, "classid":0, "index":15, "objnum":16}]
4.62963
###free gc heap memory : 496 KB

###free sys heap memory: 708 KB

Traceback (most recent call last):
  File "main.py", line 77, in <module>
ValueError: [MAIXPY]kpu: region_layer_init err!

MicroPython v0.6.2-72-g22a8555b5 on 2021-11-01; Sipeed_M1 with kendryte-k210
Type "help()" for more information.
>>> 
```

**Please complete the following information**
 - IDE version: latest (0.2.5)
 - Firmware version: `maixpy_v0.6.2_72_g22a8555b5_openmv_kmodel_v4_with_ide_support.bin`
 - Board: Sipeed MaixPy M1n
 - OS: Windows

Thanks in advance

nbenhur, 2022-09-29:
I am also getting same error [ValueError: [MAIXPY]kpu: region_layer_init err!](https://github.com/sipeed/MaixPy/issues/456#top).
IDE version: latest (0.2.5)
Firmware version: maixpy_v0.6.2_84_g22a8555b5_openmv_kmodel_v4_with_ide_support.bin
Board: Sipeed MaixPy bit-mic
OS: Linux (Ubuntu)

stnridy, 2023-11-01:
Hi anyone can help. I am seeing the same issue.

IDE version: latest (0.2.5)
Firmware version: maixpy_v0.6.2_84_g8fcd84a58_openmv_kmodel_v4_with_ide_support.bin
Board: Sipeed Maixduino
OS: Windows

Thanks in advance