sipeed/MaixPy-v1 #465: ValueError: [MAIXPY]kpu: region_layer_init err!

Norion, 2022-04-27:
**Describe the bug**
I am trying to use a selfbuilt kmodel on my Maixduino using MaixPy. I solved most compatibility errors. The kmodel has version 3.
While starting the model the program stops and yields this error message:
ValueError: [MAIXPY]kpu: region_layer_init err!

**Expected behavior**
I expected the model to load properly.

**Actual behaviour**
The program terminates without loading the kmodel

**Screenshots**
[image]

**Please complete the following information**
 - IDE version: 0.2.5
 - Firmware version: maixpy_v0.6.2_77_g11806960d_minimum_with_ide_support.bin
 - Board: Maixduino
 - OS: Windows

AIWintermuteAI, 2022-04-27:
Hi!
It might be the same issue as
https://github.com/TinkerGen/Mark-micropython/issues/1
since aXeleRate uses YOLOv3 by default and MaixPy does not support it yet (there is a PR here https://github.com/sipeed/MaixPy/pull/451 , but it is not merged)

Norion, 2022-04-27:
Oh! That might explain it, How do I get aXeleRate to use YOLOv2 instead?

Norion, 2022-04-27:
The best course for you is to train a model with legacy aXeleRate - you can do that by switching to legacy-yolov2 branch in Colab notebook right after cloning the repository.

If I understand correctly, just normally install aXeleRate and then switch branches to run the software?

AIWintermuteAI, 2022-04-27:
If you're running locally, then git clone, switch the branch and then install with ```pip install -e .```