sipeed/MaixPy-v1 #470: MaixPy image class missing most useful functions

marcusobrien, 2022-07-10:
**Describe the bug**
Maix Py **image** class is missing many useful image processing functions like find_lines()

Dir the image class and it only has these functions 

>>> import image
>>> dir(image)
['__class__', '__name__', 'ASCII', 'Image', 'ImageReader', 'ImageWriter', 'UTF8', 'font_free', 'font_load', 'get_affine_transform', 'grayscale_to_rgb', 'lab_to_rgb', 'rgb_to_grayscale', 'rgb_to_lab', 'warp_affine_ai']

The API docs mention functions like find_lines for example
_**"The line object is returned by image.find_lines, image.find_line_segments or image.get_regression.

10.1. Constructor[#](https://wiki.sipeed.com/soft/maixpy/en/api_reference/machine_vision/image/image.html#Constructor)
class image.line

Please call image.find_lines(), image.find_line_segments(), or image.get_regression() function to create this object.**_"

Using kflash_gui I installed the latest maix firmware on a Maix Bit device

maixpy_v0.6.2_83_gf0280db50_minimum_with_ide_support.bin

And when I run the code I get python error   AttributeError: 'Image' object has no attribute 'find_lines'

**To Reproduce**
Steps to reproduce the behavior:
1. Go to the example code on this page 
[(https://wiki.sipeed.com/soft/maixpy/en/api_reference/machine_vision/image/image.html)]

2. Copy and paste into IDE with Maix Bit
3. Click connect Link Icon 
4. Run program with Big green play button the code
5. AttributeError: 'Image' object has no attribute 'find_blobs'
6. The image class has lots of missing functions mentioned in the API docs

image Python module  - missing functions are mage.find_lines, image.find_line_segments and image.get_regression.

**Expected behavior**
The image class should support what is in the API docs
[mage.find_lines, image.find_line_segments or image.get_regression.](url)

**Actual behaviour**
These functions are not supported mage.find_lines, image.find_line_segments and image.get_regression.

**Screenshots**
If applicable, add screenshots to help explain your problem.

[image]

**Please complete the following information**
 - IDE version: 0.2.5
 - Firmware version: 0.6.2_83
 - Board: Maix Bit and Maix Duino
 - OS: Windows

Neutree, 2022-07-10:
[image]

marcusobrien, 2022-07-23:
Thanks @Neutree - 
Great - this worked. I didnt think I this "non-minimum" binary package would work with the IDE, as I thought only binaries named "with_ide_support" could be used with the IDE. But this "normal/full package" does include the additional code and classes, as well as work with the IDE.

closed 2022-07-23 by marcusobrien