sipeed/MaixPy-v1 #471: GC does not work perfect

sunshineharry, 2022-08-14:
**Describe the bug**
Micropython's GC does not work perfect.

**To Reproduce**
I use the code as follows
```python
class ImageDetection():
    """用摄像头进行目标检测"""
    def __init__(self, labels, anchors, model_addr, img_threshold, use_lcd = False):
        # 初始化参数
        self.input_size = (224, 224)
        self.labels, self.anchors, self.model_addr, self.img_threshold, self.use_lcd = (labels, anchors, model_addr, img_threshold, use_lcd)
        self.state = 1      # 1表示允许目标检测使用kpu，0表示禁止其使用kpu
        # 初始化相机
        sensor.reset()
        sensor.set_pixformat(sensor.RGB565)
        sensor.set_framesize(sensor.QVGA)
        sensor.set_windowing(self.input_size)
        sensor.set_hmirror(False)
        sensor.set_vflip(False)
        sensor.run(1)
        # 初始化LCD
        if self.use_lcd:
            lcd.init(type=1)
            lcd.rotation(0)
            lcd.clear(lcd.WHITE)
        # 初始化kpu
        gc.collect()
        self.task = kpu.load(self.model_addr)
        kpu.init_yolo2(self.task, self.img_threshold, 0.3, 5, self.anchors)

    def run_detection(self):
        """运行目标检测"""
        img = sensor.snapshot()
        t = time.ticks_ms()
        objects = kpu.run_yolo2(self.task, img)
        t = time.ticks_ms() - t
        if objects:
            for obj in objects:
                pos = obj.rect()
                img.draw_rectangle(pos)
                img.draw_string(pos[0], pos[1], "%s : %.2f" %(labels[obj.classid()], obj.value()), scale=2, color=(255, 0, 0))
        img.draw_string(0, 200, "t:%dms" %(t), scale=2, color=(255, 0, 0))
        if self.use_lcd:
            lcd.display(img)

    def deinit(self):
        """释放kpu内存"""
        if not self.task is None:
            kpu.deinit(self.task)

if __name__ == "__main__":
    # 初始化并运行目标检测
    labels = ['ball']
    anchors = [1.38, 1.31, 2.84, 2.75, 2.47, 2.47, 2.19, 2.06, 3.19, 3.16]
    img_detection = ImageDetection(labels, anchors, model_addr=0x240000, img_threshold=0.5, use_lcd=True)
    try:
        while True:
            img_detection.run_detection()
            comm.send(str(i))
            print(gc.mem_free()) 
    except Exception as e:
        raise e
    finally:
        img_detection.deinit()
```

**Expected behavior**
The GC works properly to ensure that the memory is not consumed.

**Actual behaviour**
The `print(gc.mem_free())` shows that the memory will be out of use. After one miniute, `MaixPy IDE` cannot communicate to MAIX DOCK.
```
457280
456704
456128
455552
454976
454400
453824
453248
452672
452096
```

Then, I add some code as follows:
```python
# 垃圾回收，使用定时器触发，每10s执行一次
def __gc_collect(timer):
    gc.collect()
    print(gc.mem_free())

gc_collecter = Timer(Timer.TIMER0, Timer.CHANNEL0, mode=Timer.MODE_PERIODIC, period=10, unit=Timer.UNIT_S, callback=__gc_collect, arg=__gc_collect, start=False, priority=1, div=0)
gc_collecter.start()
```
The code means after 10 seconds, GC will be work once. Then, the memory will not be out of use.

**Please complete the following information**
 - IDE version: 0.2.5
 - Firmware version: maixpy_v0.6.2_83_gf0280db50_openmv_kmodel_v4_with_ide_support
 - Board: Maix Dock
 - OS: Windows

robert-hh, 2022-08-14:
Since gc.collect() does not move blocks when collecting trash, memory fragmentation cannot be avoided in all cases. And so it may happen that a memory area cannot be allocated eve if the total amount of free RAM is large. 
The approach you made of regular and early call GC.collect() is a good strategy to keep the heap free.
There have been GC changes in mainline MicroPython. These may help to improve memory handling.

sunshineharry, 2022-08-15:
thank you very much.