sipeed/MaixPy-v1 #486: 更新子模块时出错

BitaMatt, 2023-05-24:
**Describe the bug**

在输入git submodule update --recursive --init后出现下面错误

fatal: Fetched in submodule path 'tools/flash/kflash_py', but it did not contain 1ef6f4c0b2cb8b1872b6ffe9337f4e02d5487fa6. Direct fetching of that commit failed.

查询了一下，说是子仓库kflash_py没有这个哈希码的提交版本，所以获取资源失败。请问是不是这个子模块的更新是因为作者没同步好才导致的？

**Screenshots**
[image]

**Please complete the following information**
 - OS: WSL Ubuntu22.04 LTS

vowstar, 2023-05-24:
我clone了master分支下来，是正常的。
Submodule path 'tools/flash/kflash_py': checked out `'550828c768b16ef329695d3f5eace3f6bcf14af2'`
估计你所在分支的submodule是坏的，而对于kflash, 推荐你切到 `'550828c768b16ef329695d3f5eace3f6bcf14af2'`

BitaMatt, 2023-05-25:
> 

我是按照build.md的步骤去克隆的，但就是出现这个错误

BitaMatt, 2023-05-25:
> > 
> 
> 我是按照build.md的步骤去克隆的，但就是出现这个错误

不知道为什么，今天把Maixpy文件全删除了，然后重新克隆就成功了！没再出现这个问题了