sipeed/MaixPy-v1 #491: MAixPy IDE + MAix AMIGO   y i e l d i n g   ImportError: cannot import name MIC_ARRAY

NameStream, 2023-09-17:
su09-17-2023

ImportError: cannot import name MIC_ARRAY 

MAix Amigo

(1.) Hi, there.

(2.) Using MAixPy IDE, receiving error code for MAix Amigo:  

ImportError: cannot import name MIC_ARRAY 

(3.) MAix AMIGO 

(3a.) SP-MOD MicArray Module

(3b.) Connector:  10P 0.5mm FPC

(3c.) Sipeed MAix R6 + 1 Microphone Array Board

(3d.) Located inside the back of the developer board #2970, moved the switch to M1N instead of STM.  
* * * * * * * * {  By the way, STM yields same results. } * * * * * * * *

(3e.) MAixPy IDE:  Tools > Select Board > Sipeed Maixduino

(3f.) Will only run the code for “helloworld.py”, although the LCD image orientation is misaligned, upside down; the picture is smaller than the LCD screen; but image has mostly clear resolution without any noticeable lag, latency, delay.

(3g.) Receiving “ImportError:  cannot import name MIC_ARRAY”
when attempting to run the code for:  

ROUTINE Sound Source Localization

https://wiki.sipeed.com/soft/maixpy/en/develop_kit_board/module_microphone.html 

*******and*******also*******for*******

AirPocket Soundman’s Sound Localizer
https://github.com/airpocket-soundman/SoundVisualizer/blob/main/src/mic_array_en_ver204.py

(3h.) Pin assignments used for MAix AMIGO

LED_CLK_GPIOHS28…….23

LED_DAT_GPIOHS27…….20

DAT 0…………….......……12

DAT 1……………..…..…….…9

DAT 2………………..…...……31

DAT 3………….……..……...7

WS……………..…....…..……19

BCK_SCK……...…..…………17

(4.) Would you list the correct pin assignments for the:

(4a.) MAix Amigo #2970 in the M1n position

(4b.) MAix Amigo #2970 in the STM position

(5.) Thanks.

(6.) Toodles!

silenzio777, 2023-11-01:
1 Nov 2023

MAixPy IDE, board Maix Bit (with Mic)
Sipeed R6+1 Microphone Array Module

Please help. The same error:
"ImportError: cannot import name MIC_ARRAY"

Where module MIC_ARRAY is located?

Thanks!

silenzio777, 2023-11-01:
> 1 Nov 2023
> MAixPy IDE, board Maix Bit (with Mic) Sipeed R6+1 Microphone Array Module
> Please help. The same error: "ImportError: cannot import name MIC_ARRAY"
> Where module MIC_ARRAY is located? 
> Thanks!

Solved.
Flash board with "maixpy_v0.6.2_85_g23d09fbcc_with_lvgl.bin" file.
kflash -p "/dev/YOUR_DEVICE_NAME" -b 1500000 -B bit_mic maixpy_v0.6.2_85_g23d09fbcc_with_lvgl.bin

From here - https://dl.sipeed.com/shareURL/MAIX/MaixPy/release/master/maixpy_v0.6.2_85_g23d09fbcc